Xilinx/BNN-PYNQ #111: 硬件设计重建的问题

ruibao123, 2019-08-23:
在win10系统上安装的VirtualBox虚拟机ubuntu

在虚拟机上安装的vivado2018.2 环境变量设置为 <clone_path>/BNN_PYNQ/bnn/src/

在虚拟机上安装github   终端输入 sudo apt-get install git

然后终端输入 ./make-hw.sh cnvW1A1 pynqZ1-Z2 h

之后运行后报错

rui@rui-VirtualBox:~/BNN_PYNQ/bnn/src/network$ ./make-hw.sh cnvW1A1 pynqZ1-Z2 h
xilinx-tiny-cnn already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'rui' on host 'rui-VirtualBox' (Linux_x86_64 version 5.0.0-25-generic) on Fri Aug 23 17:57:34 CST 2019
INFO: [HLS 200-10] On os Ubuntu 18.04.2 LTS
INFO: [HLS 200-10] In directory '/home/rui/BNN_PYNQ/bnn/src/network/output/hls-syn'
HLS project: cnvW1A1-pynqZ1-Z2
HW source dir: /home/rui/BNN_PYNQ/bnn/src/network/cnvW1A1/hw
BNN HLS library: /home/rui/BNN_PYNQ/bnn/src/library/hls
INFO: [HLS 200-10] Creating and opening project '/home/rui/BNN_PYNQ/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/rui/BNN_PYNQ/bnn/src/network/cnvW1A1/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/rui/BNN_PYNQ/bnn/src/network/cnvW1A1/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/rui/BNN_PYNQ/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/rui/BNN_PYNQ/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Creating and opening solution '/home/rui/BNN_PYNQ/bnn/src/network/output/hls-syn/cnvW1A1-pynqZ1-Z2/sol1'.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Fri Aug 23 17:57:35 2019...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

giuliogamba, 2019-09-17:
Can you please open the Vivado HLS GUI and re-run csim? It might give more details about the error.

bibizxd, 2019-10-22:
You can retry in Ubuntu 16.04.

closed 2020-09-30 by giuliogamba