Xilinx/BNN-PYNQ #113: Jupyter Notebook: OSError - Cannot load library

jstran97, 2019-09-21:
Hi! So I've been having this issue where I can't run this set of code on Jupyter Notebook. 
I currently have both the PYNQ Z1 and Z2 in my possession, but I got this error while trying to run that code using the Z1. 
I also have the PYNQ Z1 image ver 2.4

I'm not sure what the cause of this error is, but does it have to do with how the commands 
"sudo pip3 install git+https://github.com/maltanar/BNN-PYNQ.git", 
"sudo pip3 install --upgrade git+https://github.com/maltanar/BNN-PYNQ.git", 
"sudo -H pip3 install --upgrade git+https://github.com/maltanar/BNN-PYNQ.git", and
"sudo pip3 install git+ssh://git@github.com/maltanar/BNN-PYNQ.git" all gave me the same error code 128?
Since all of those pip3 install commands failed, I downloaded the .zip file from github.com/maltanar/BNN-PYNQ, extracted it, and used Samba to copy it and store it into the PYNQ directories. 

[image]
[image]

I'm really sorry for this long post, but I've looked into nearly all of the possible errors and solutions out there. Sadly, none of them has worked so far.

jstran97, 2019-09-24:
The output I should be getting for 

"import bnn
root_dir = bnn.bnn.BNN_ROOT_DIR
param_dir = bnn.bnn.BNN_PARAM_DIR
print(root_dir)
print(param_dir)"   in Jupyter Notebook is 

/usr/local/lib/python3.6/dist-packages/bnn
/usr/local/lib/python3.6/dist-packages/bnn/params

The command "sudo pip3 install git+https://github.com/maltanar/BNN-PYNQ.git" and "sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git" did not work for me in the beginning. 

[image]

I think because of those 2 commands failing, the directory "bnn" does not exist in my /usr/local/lib/python3.6/dist-packages. Therefore, I think this issue led up to the OS Error above ^.

giuliogamba, 2020-04-28:
When you install this repository, do the other examples work? I mean MNIST and Cifar10.

closed 2020-09-30 by giuliogamba

kavyakl, 2021-07-06:
Hello, 
How did you resolve this error?
I am getting the same error.
Any help will be appreciated.
Thank you