Xilinx/BNN-PYNQ #114: ModuleNotFoundError: No module named 'bnn'

minertom, 2019-09-29:
Hi,,
I am using the BNN-PYNQ demo on the ultra96V2 board. I'm running a jupyter notebook on the board. The BNN-PYNQ was sucessfully loaded.
However, when attempting to "import bnn"  I get 

Would you recommend the ## Hardware design rebuilt as discussed in the README?

---------------------------------------------------------------------------
ModuleNotFoundError                       Traceback (most recent call last)
<ipython-input-5-6f5bcc2924e3> in <module>()
----> 1 import bnn

ModuleNotFoundError: No module named 'bnn'

giuliogamba, 2019-10-10:
Can you give more details about the failure of the `sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git` command? 
Another option is indeed to clone the repo and after that executing `sudo pip3 install ./` in the main folder. 
Does that solved your problem?

minertom, 2019-10-13:
Hi giuliobamba,

I was just able to find your reply, and I thank you for it. I'm recovering from the combined hellscape of having no power here in Northern California for a couple of days, and then having a resulting disk failure. 

1) As best as I remember, the suggested method of installation, `sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git` failed with an error something to the effect that the receiver, meaning git, did not understand the command and there was like an error 128. I can't boot from that disk anymore so I can't just run the command at the moment.

2) I used `git clone https://github.com/Xilinx/BNN-PYNQ.git` in order to clone (but there were still problems). I did not do any kind of "installation" step after the clone.

3) In your post you said 

> clone the repo and after that executing sudo pip3 install ./ in the main folder.

How much different is your version of the command for cloning from my version? Also, could you please be more specific about the command "sudo pip3 install ./" ? I use pip3 to install packages but in this case what comes after the ./ ?

Thank you again.
Tom

giuliogamba, 2019-10-14:
Hi, 

the command `pip3` is actually executing automatically the python package installation by executing the file `setup.py`. If you clone the complete repo locally on your PYNQ board, change directory to the main `BNN-PYNQ` and execute `sudo pip3 install ./` it will automatically look for the `setup.py` file and execute the installation in the same way as by executing `sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git`.
Hope this helps

giuliogamba, 2019-11-21:
@minertom Does the last post solved your problem?

closed 2019-12-06 by giuliogamba