Xilinx/BNN-PYNQ #115: Regarding the "#if defined(OFFLOAD) && defined(RAWHLS)", could you please tell me how to define the variable "RAWHLS" ?

HonggangYu, 2019-10-07:
Hi,

I want to compile "BlackBoxJam" function in the file "BNN-PYNQ/bnn/src/library/host/foldedmv-offload.h", and find that the variable "RAWHLS" in the #134 line {#if defined(OFFLOAD) && defined(RAWHLS)} need to be predefined. However, when I go through the whole source code provided by Xilinx (https://github.com/Xilinx/BNN-PYNQ), only the variable "OFFLOAD" was predefined in the #58 line of the file "BNN-PYNQPYNQ/bnn/src/network/cnvW1A1/sw/main_python.cpp".

Could you please tell me how to define the variable "RAWHLS" ?   

Could you suggest how to fix this issue?

giuliogamba, 2019-10-10:
I don't think I understand the problem. 
Anyway, those variable are defined at compile time according to the need in the [make-sw.sh](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-sw.sh) script. 
When we compile the code using software execution (thus compiling the HLS C++ code as software), we use the `-DRAWHLS` (line 108) flag to compile that part of the code.

closed 2019-10-10 by giuliogamba

HonggangYu, 2019-10-17:
> I don't think I understand the problem.
> Anyway, those variable are defined at compile time according to the need in the [...]

Hi Giulio, 

Thanks for your reply! Could you please introduce the difference between the code using software execution and the code using hardware execution? 

Thanks!