Xilinx/BNN-PYNQ #120: Errors in Hardware Rebuilding

imhjnju, 2019-10-23:
When I cloned the repository, the **finn-hlslib** was not cloned together, as a result, error appeared, with message "**bnn_library.h was not found**",   

Then, I coloned the  [finn-hlslib](https://github.com/Xilinx/finn-hlslib/) . I tried 2 versions, one was current version, another was with [SHA:b5dc957](https://github.com/Xilinx/finn-hlslib/tree/b5dc957a16017b8356a7010144b0a4e2f8cfd124), which was redirected by the folder in this repository. Both arised the same error:  

**In file included from ../../../../../../cnvW2A2/hw/top.cpp:45:
In file included from .../BNN-PYNQ/bnn/src/library/finn-hlslib/bnn-library.h:62:
..../BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:107:3: error: no matching function for call to 'Matrix_Vector_Activate_Batch'
  Matrix_Vector_Activate_Batch<MatrixW, MatrixH, SIMD, PE, TSrcI, TDstI, TWeightI>**

_Thank You for your help :-)_

giuliogamba, 2019-10-23:
Hi, 

can you please try to clone again the BNN-PYNQ repo using the command: 
`git clone https://github.com/Xilinx/BNN-PYNQ.git --recursive`
and retry the hardware generation? 
I cannot replicate your error

njuhj, 2019-10-23:
Thank you for your help :-) 
I ignored  the '--recursive'....

imhjnju, 2019-10-23:
Thank You for your help :-)
I ignored the **'--recursive'**...

giuliogamba, 2019-11-19:
@imhjnju Is this problem solved? if yes, please clone the issue

closed 2019-11-22 by imhjnju