Xilinx/BNN-PYNQ #121: Both pip3 and pip3.6 install not working[PYNQ-Z1]

aanujdu, 2019-10-23:
My setup: PYNQ-Z1 board with terminal opened from Jupyter notebook
I just upgraded to the recent pynq version by executing:
`sudo pip3.6 install --upgrade git+https://github.com/Xilinx/PYNQ.git`
Now I am trying to install the BNN-PYNQ framework on the system, but both pip3 and pip3.6 installations fail with the following log:
```
root@pynq:/home/xilinx# sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git
Downloading/unpacking git+https://github.com/Xilinx/BNN-PYNQ.git
  Cloning https://github.com/Xilinx/BNN-PYNQ.git to /tmp/pip-9nq7p508-build
  Running setup.py (path:/tmp/pip-9nq7p508-build/setup.py) egg_info for package from git+https://github.com/Xilinx/BNN-PYNQ.git
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/tmp/pip-9nq7p508-build/setup.py", line 34, in <module>
        import bnn
      File "/tmp/pip-9nq7p508-build/bnn/__init__.py", line 1, in <module>
        from .bnn import PynqBNN, CnvClassifier, LfcClassifier, RUNTIME_HW, RUNTIME_SW
      File "/tmp/pip-9nq7p508-build/bnn/bnn.py", line 30, in <module>
        from pynq import Overlay, PL
    ImportError: No module named 'pynq'
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/tmp/pip-9nq7p508-build/setup.py", line 34, in <module>

    import bnn

  File "/tmp/pip-9nq7p508-build/bnn/__init__.py", line 1, in <module>

    from .bnn import PynqBNN, CnvClassifier, LfcClassifier, RUNTIME_HW, RUNTIME_SW

  File "/tmp/pip-9nq7p508-build/bnn/bnn.py", line 30, in <module>

    from pynq import Overlay, PL

ImportError: No module named 'pynq'
----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-9nq7p508-build
Storing debug log for failure in /root/.pip/pip.log
```
**pip3.6 installation**
```
root@pynq:/home/xilinx# sudo pip3.6 install git+https://github.com/Xilinx/BNN-PYNQ.git
Collecting git+https://github.com/Xilinx/BNN-PYNQ.git
  Cloning https://github.com/Xilinx/BNN-PYNQ.git to /tmp/pip-ecrrahsj-build
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-ecrrahsj-build/setup.py", line 34, in <module>
        import bnn
      File "/tmp/pip-ecrrahsj-build/bnn/__init__.py", line 1, in <module>
        from .bnn import PynqBNN, CnvClassifier, LfcClassifier, RUNTIME_HW, RUNTIME_SW
      File "/tmp/pip-ecrrahsj-build/bnn/bnn.py", line 30, in <module>
        from pynq import Overlay, PL
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/__init__.py", line 33, in <module>
        from .pl import PL
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/pl.py", line 39, in <module>
        from .ps import CPU_ARCH_IS_SUPPORTED, CPU_ARCH, ZYNQ_ARCH, ZU_ARCH
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/ps.py", line 322, in <module>
        class _ClocksUltrascale(_ClocksMeta):
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/ps.py", line 382, in _ClocksUltrascale
        IOPLL_CTRL = Register(CRL_APB_ADDRESS + IOPLL_CTRL_OFFSET)
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/registers.py", line 115, in __init__
        array = MMIO(address, np.dtype(register_type).itemsize).array
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/mmio.py", line 82, in __init__
        device = Device.active_device
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/pl_server/device.py", line 90, in active_device
        if len(cls.devices) == 0:
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/pl_server/device.py", line 78, in devices
        cls._devices.extend(DeviceMeta._subclasses[key]._probe_())
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/pl_server/device.py", line 526, in _probe_
        return [XlnkDevice()]
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/pl_server/device.py", line 535, in __init__
        self.default_memory = Xlnk()
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/xlnk.py", line 136, in __init__
        Xlnk._open_library(Xlnk.libxlnk_path)
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/xlnk.py", line 122, in _open_library
        cls.libxlnk = cls.ffi.dlopen(path)
      File "/opt/python3.6/lib/python3.6/site-packages/cffi/api.py", line 146, in dlopen
        lib, function_cache = _make_ffi_library(self, name, flags)
      File "/opt/python3.6/lib/python3.6/site-packages/cffi/api.py", line 828, in _make_ffi_library
        backendlib = _load_backend_lib(backend, libname, flags)
      File "/opt/python3.6/lib/python3.6/site-packages/cffi/api.py", line 823, in _load_backend_lib
        raise OSError(msg)
    OSError: cannot load library '/usr/lib/libcma.so': /usr/lib/libcma.so: cannot open shared object file: No such file or directory.  Additiona
lly, ctypes.util.find_library() did not manage to locate a library called '/usr/lib/libcma.so'
    Exception ignored in: <bound method Xlnk.__del__ of <pynq.xlnk.Xlnk object at 0x34ec94f0>>
    Traceback (most recent call last):
      File "/opt/python3.6/lib/python3.6/site-packages/pynq/xlnk.py", line 150, in __del__
    AttributeError: 'Xlnk' object has no attribute 'bufmap'

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-ecrrahsj-build/
You are using pip version 9.0.1, however version 19.3.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```
1. What  could be the issue? Should I upgrade pip as suggested in the error log?
2. Also how can I find what version of pynq is currently installed on my board?

Thanks!

giuliogamba, 2019-10-23:
I suggest to upgrade to PYNQ 2.5 by downloading the image [here](http://www.pynq.io/board.html). You can check your current PYNQ version on the terminal with `pip3 show pynq` or by checking the file `~/REVISION`.

closed 2019-10-23 by giuliogamba

mehdi-najib, 2021-06-09:
♥