Xilinx/BNN-PYNQ #125: Varying PEs and SIMDs parameters

marco-pag, 2019-12-12:
Hello,
We are doing some tests varying the number of PEs/SIMDs parameters in each layer of the cnvW1A1 (CIFAR10) network to improve its response time. Everything works fine, and the response times of each layer scales accordingly to the number of PE/SIMD except for layer 0. If we modify the number of PEs/SIMDs in layer 0 away from the default (that is 16/3), the network produces incorrect results. Should we also change the comparison type (i.e., ap_fixed<24, 16>) or some other parameters in order to make layer 0 working correctly?

giuliogamba, 2019-12-13:
Which set of values have you tried for layer0? Which kind of error do you see? Compilation or just misclassification?

marco-pag, 2019-12-13:
We have tried with (PEs = 32, SIMDs = 3), and (PEs = 64, SIMDs = 3). In both cases, we get misclassification but no compilation error. The code compiles correctly. We decided to keep SIMDs = 3 because it seems somehow linked to the number of input (color) components. Is this reasoning correct? Or, on the contrary, we can change the SIMDs number to an arbitrary value like say 10?

giuliogamba, 2019-12-16:
This is extremely odd, I've just tried with the latest commit to generate the SIMD=3 and PE=32 for the first layer and everything went through. Can you try with the attached generated files? 
[binparam-cnvW1A1-pynq.zip](https://github.com/Xilinx/BNN-PYNQ/files/3968705/binparam-cnvW1A1-pynq.zip)

marco-pag, 2019-12-17:
We have tried with the weight you send me, but we still get a misclassification. I have to say, though, that we are embedding the weights and the thresholds directly into the accelerator (at HLS compile time) using the header files generated by the "cifar10-gen-weights-W1A1.py" script for performance reasons. Here (https://hastebin.com/xaheniwana.c), I've uploaded a text file containing both i) the thresholds generated using the python script and ii) the thresholds extracted from the binary file you send me. Considering the two's complement representation, they are the same. So, we don't understand why we get a misclassification error. When changing the PEs and SIMDs of layer 0, do we also need to change any of the parameters of the convolutional layer (except the parameters coming from config.h)?

`ConvLayer_Batch<L0_K, L0_IFM_CH, L0_IFM_DIM, L0_OFM_CH, L0_OFM_DIM, L0_SIMD, L0_PE, Slice<ap_fixed<8, 1, AP_TRN, AP_SAT>>, Identity, Recast<Binary>>(inter0_2, inter1, weights0, threshs0, NUM_REPS, ap_resource_lut());`

giuliogamba, 2019-12-20:
Have you tried with those weights and changed the config.h file in the cnvw1a1/hw folder? You don't need to change anything else, and the fact that you get misclassification while for me it works it's odd. Do you have your code forked somewhere in a repo?

closed 2020-09-30 by giuliogamba