Xilinx/BNN-PYNQ #127: BNN-PyNQ on ZCU-104

k4mils, 2019-12-12:
Hello. Is it possible to create bitstream for the ZCU-104 board with one of those cnv networks from the repository?

giuliogamba, 2019-12-13:
Yes, it is. I suggest to start from the project targeting the Ultra96 board and adapt the scripts available [here](https://github.com/Xilinx/BNN-PYNQ/tree/master/bnn/src/library/script) to target the ZCU104.

k4mils, 2019-12-16:
Thank you for the answer. I've found the following dependencies and have some questions:

- bnn/libraries/ultra96/
There are .so files named *ultra96.so. Are they specific to the platform and have to be swapped(how?) or can I leave them as they are

- bnn/src/library/driver/platform-xlnk.cpp
There are ADDR_REG_BASE defines - which addres base value is appropriate for ZCU104?

- bnn/src/library/script/ultra96/make-vivado-proj.tcl
Is it enough to swap the part name here? (in set_proj_part)

- bnn/src/library/script/ultra96/ultra96.tcl
Where can I get a proper file for ZCU104 from?

- bnn/src/library/script/ultra96/ultra96.xdc
I found the .xdc file for ZCU on XILINX page, is it ok to just swap it with the one for ultra96?

- bnn/src/network/make-hw.sh
I've found the part name again as well as the target clock. Is the clock somehow constrained for the board or can I use the same as for ultra96?

giuliogamba, 2019-12-17:
Hi, 
here some answers: 

> Thank you for the answer. I've found the following dependencies and have some questions:
> 
> * bnn/libraries/ultra96/
>   There are .so files named *ultra96.so. Are they specific to the platform and have to be swapped(how?) or can I leave them as they are

You need to recompile on board (ZCU104) using the same compile scripts as for ultra96. 
> * bnn/src/library/driver/platform-xlnk.cpp
>   There are ADDR_REG_BASE defines - which addres base value is appropriate for ZCU104?

The same as Ultra96
> * bnn/src/library/script/ultra96/make-vivado-proj.tcl
>   Is it enough to swap the part name here? (in set_proj_part)

I would not run the scripts, but manually do a new vivado project and add the HLS generated IP. 
> * bnn/src/library/script/ultra96/ultra96.tcl
>   Where can I get a proper file for ZCU104 from?
> * bnn/src/library/script/ultra96/ultra96.xdc
>   I found the .xdc file for ZCU on XILINX page, is it ok to just swap it with the one for ultra96?

There's a board support package for ZCU104 in Vivado, you should use that. 
> * bnn/src/network/make-hw.sh
>   I've found the part name again as well as the target clock. Is the clock somehow constrained for 
the board or can I use the same as for ultra96?

Same clock as for Ultra96 should work.

closed 2019-12-17 by giuliogamba