Xilinx/BNN-PYNQ #131: Building the project in VHDL instead of Verilog

MuBadawy, 2020-01-16:
Hey, during rebuilding the project in vivado, the project generated files are in verilog. Is there anyway to get these generated files in VHDL instead. I tried to change the target language after building the project but it looks like it's not a good idea. I am beginner so I'd appreciate if there is a way to achieve that. 

Thanks,

giuliogamba, 2020-01-20:
What are you trying to achieve? The Vivado HLS project generates both Verilog and VHDL, you can check the generated code under `src/network/output/hls-syn/"your_network"/sol1/impl`

MuBadawy, 2020-01-23:
I was trying to have the VHDL files in the Vivado project (xpr file) itself to work on them and manipulate them. I have another question though, whicj HDL files are respoonsible for the XNOR operation between the weights and the activations?

giuliogamba, 2020-04-28:
Within the Vivado HLS GUI, you can open the project, move on the Analysis pane and you can check the naming given to all variables after the HLS synthesis, and you that as a reference to check the portion of HDL code you want to analyze.