Xilinx/BNN-PYNQ #133: Performance Information on current BNN-PYNQ version

vqhieu138, 2020-04-03:
Hello team,
I cloned BNN-PYNQ.git and ran well with "Hardware rebuilt -lfcW1A1-ultra96 (vivado 2018.2)" following your guideline. I checked Vivado report, and there are some questions, hope you can help me:
1. Result on current lfcW1A1-ultra96 is different from ""FINN-R: An End-to-End Deep-Learning Framework for Fast Exploration of Quantized Neural Networks". So can you show me where I can get performance information for the current version? such as : FPS (frame per second), GOP/s, GOP/s/W, accuracy ..?
2. If you haven't measured performance for the current lfc1W1A1-ultra96, Can you show me where I can get the guideline to rebuild Vivado project being similar version with "FINN-R" paper?
(The performance reuslt on FINN-R paper: 
FINN-R MLP-4 --- Ultra96(DF) --300Mhz -- 417 LUTs --- 5110 (75%) GOp/s --- 11.8W -- 433 GOp/s/W

Thanks

giuliogamba, 2020-04-05:
Can you please give more details about the mismatch in performances you experienced? In order to measure the framerate and accuracy you can execute the notebook [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/notebooks/LFC-QNN_MNIST.ipynb). Then you can count the ops of the network (6 MOps as in table 4 of [FINN-R](https://arxiv.org/pdf/1809.04570.pdf)) and you multiply by the framerate in order to get the thoughput.

vqhieu138, 2020-04-06:
Hello,

Thank you for your reply, after running command:
"./make-hw.sh lfcW1A1 ultra96 a"
I can open vivado project and based on vivado report, I have the following information:
26809 9(LUTs) --- 30947(FFs) ----110 (BRAM) --- 1731 (LUTRAM)
Meanwhile, on FINN-R paper, for FINN-R_MLP 4:
38205(LUTs) ----417(BRAM18).
That's the reason why I thought that the current design version was different from the version on FINN-R paper.
Regarding measurement, thank you so much for your guideline, but, I haven't had ultra96 so far, I had only zcu102 board, can I convert the design to adapt to this board instead of ultra96?

giuliogamba, 2020-04-28:
Hi, 

unfortunately the hardware usage reported in the paper was based on an internal release, so in this repository there isn't the exact same hardware cost. On the other hand, I will strongly suggest to use the latest code available, since in time we performed a lot of optimizations to improve the quality of results (hardware cost) while having the same performances. 
In terms of porting to a different board like 102, from the hardware point of view there should be no problem, but you will need to change the target platform [here ](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-hw.sh#L135) and generate the vivado block design by hand. From the host integration point of view, this repository relies on PYNQ as a platform, there is work in the community on how to port PYNQ to ZCU102 like [this guide](http://www.programmersought.com/article/1965258506/).

vqhieu138, 2020-05-26:
Hello ,

Thank you very much for your answer. I had desirable information. It is really great if performance is maintained, while getting hardware code reduction. (LUTs from 38k --> 26K).

closed 2020-09-30 by giuliogamba