Xilinx/BNN-PYNQ #136: Change the dimension of Input

1430533056, 2020-04-26:
Hello,
I have trained a new network with input dimension 188x188x3,  all params, .bit file and .so file are generated. The problem is that the accuracy in pynq is much slow than the training phrase in theano. I guess the reason is the change of the input dimension. To apply the changes, I replace 32 with 188 in line 163,168,169 in file [top.cpp](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/hw/top.cpp) and in line 59, 60 in [main_python.cpp.](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/sw/main_python.cpp). I wonder whether I should make changes in other places ?

giuliogamba, 2020-04-28:
Hi, 

if you changed the input dimensions, most likely all the intermediate dimensions will also change. Those are listed in the `config.h ` file as `IFM_DIM ` and `OFM_DIM` as in the example [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/hw/config.h#L43). 
That file is automatically generated from the export script [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/training/cifar10-gen-weights-W1A1.py), line 42 and 43. Those have to be adjusted according to the topology you trained with the new input dimension.

1430533056, 2020-04-28:
I have adjusted the topology in gtsrb-gen-binary-weights.py  and the config.h is automatically generated. But it still has a very low accuracy. Besides the above files, any other files should I make changes? By the way, does the ConvLayer support padding ?

closed 2020-05-02 by 1430533056