Xilinx/BNN-PYNQ #138: 'bnn-library.h' file not found

hwtsnana, 2020-06-01:
Hi,  I am trying to rebuild the hardware design with ./make-hw.sh cnvW1A2 pynqZ1-Z2 a. but I'm getting this error. please help
My os is Ubuntu 16.04.6 with Vivado 2018.2
hewa@hewa-VB:~/Downloads/BNN-PYNQ/bnn/src/network$ source /opt/Xilinx/Vivado/2018.2/settings64.sh
hewa@hewa-VB:~/Downloads/BNN-PYNQ/bnn/src/network$ export XILINX_BNN_ROOT=~/Downloads/BNN-PYNQ/bnn/src/
hewa@hewa-VB:~/Downloads/BNN-PYNQ/bnn/src/network$ export VIVADOHLS_INCLUDE_PATH=/opt/Xilinx/Vivado/2018.2/include/
hewa@hewa-VB:~/Downloads/BNN-PYNQ/bnn/src/network$ ./make-hw.sh cnvW1A2 pynqZ1-Z2 a
xilinx-tiny-cnn already cloned
FINN hls library already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.2 (64-bit)
  **** SW Build 2258646 on Thu Jun 14 20:02:38 MDT 2018
  **** IP Build 2256618 on Thu Jun 14 22:10:49 MDT 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /opt/Xilinx/Vivado/2018.2/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/opt/Xilinx/Vivado/2018.2/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'hewa' on host 'hewa-VB' (Linux_x86_64 version 4.15.0-101-generic) on Mon Jun 01 12:53:01 NZST 2020
INFO: [HLS 200-10] On os Ubuntu 16.04.6 LTS
INFO: [HLS 200-10] In directory '/home/hewa/Downloads/BNN-PYNQ/bnn/src/network/output/hls-syn'
HLS project: cnvW1A2-pynqZ1-Z2
HW source dir: /home/hewa/Downloads/BNN-PYNQ/bnn/src//network/cnvW1A2/hw
BNN HLS library: /home/hewa/Downloads/BNN-PYNQ/bnn/src//library/finn-hlslib
INFO: [HLS 200-10] Opening project '/home/hewa/Downloads/BNN-PYNQ/bnn/src/network/output/hls-syn/cnvW1A2-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/hewa/Downloads/BNN-PYNQ/bnn/src//network/cnvW1A2/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/hewa/Downloads/BNN-PYNQ/bnn/src//network/cnvW1A2/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/hewa/Downloads/BNN-PYNQ/bnn/src//library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/hewa/Downloads/BNN-PYNQ/bnn/src//library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/hewa/Downloads/BNN-PYNQ/bnn/src/network/output/hls-syn/cnvW1A2-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
   Compiling ../../../../../../../library/host/rawhls-offload.cpp in debug mode
csim.mk:86: recipe for target 'obj/rawhls-offload.o' failed
In file included from ../../../../../../../library/host/rawhls-offload.cpp:44:
/home/hewa/Downloads/BNN-PYNQ/bnn/src//library/host/foldedmv-offload.h:136:10: fatal error: 'bnn-library.h' file not found
#include "bnn-library.h"
         ^
1 error generated.
make: *** [obj/rawhls-offload.o] Error 1
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Mon Jun  1 12:53:07 2020...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

giuliogamba, 2020-06-02:
Have you set the environmental variable `XILINX_BNN_ROOT` as per point 3 in the [readme ](https://github.com/Xilinx/BNN-PYNQ/blob/master/README.md) file?

hwtsnana, 2020-06-02:
Hi @giuliogamba 
I didn't use the `--recursive` flag.
I will try with the flag.
Thanks for the reply.

hwtsnana, 2020-06-03:
It worked...
Thank you @giuliogamba

closed 2020-06-03 by hwtsnana