Xilinx/BNN-PYNQ #139: how do we create our own topology?

Adel-Chaker, 2020-06-24:
Morning, everyone,

I am a student, and I am looking to test a topology other than the one provided (BNN, CFL).

I'm sorry, I saw that there were a lot of open topics on the same problem. 
but is it possible that someone can clarify all the steps?

Thank you very much.

giuliogamba, 2020-06-30:
You might want to check FINN project [here](https://github.com/Xilinx/finn) for training your own topology.

Adel-Chaker, 2020-06-30:
Hello, 

Thank you. I'll look into it.

But first I'd like to check my idea if it's possible, here's what I've done so far:

1- I've created a new topology (looks like CNV but with two extra 2D convolution layers at the input) with 5x5 filters for a 40x40 input image.

2- I created a python script to generate a dataset from images.

3- I created a python script to run a training from my dataset on my new topology.

4- I created a python script to generate my weights and the config.h file for my topolohy.

now I'm at the "top.cpp" step, I edited the "DoMemInit" function to add my two layers (case Conv and Case threshs).

but I don't understand the "DoCompute" function, and what should I edit in it?

I forgot a step?

PS: I have already done a test with my dataset on the CNV and it seems to work perfectly. The problem here is to create an original topology.