Xilinx/BNN-PYNQ #140: Where is the fold "10_mnist_pictures"

yyyzzzyyy, 2020-07-27:
Hello, I am trying to use the code of LFC-QNN_MNIST.ipynb. However, when I try to execute the code:

"with open("/home/xilinx/jupyter_notebooks/bnn/10_mnist_pictures", "wb") as out_file:
    with open("/home/xilinx/jupyter_notebooks/bnn/t10k-images-idx3-ubyte","rb") as img_file:"

It throw the error of:
"FileNotFoundError: [Errno 2] No such file or directory: '/home/xilinx/jupyter_notebooks/bnn/10_mnist_pictures".

My question is where is the fold "10_mnist_pictures"?
Thank you.

yangliuxi007, 2020-07-27:
I'm don't know, bro. sorry......

giuliogamba, 2020-07-27:
That's a file which is generated in the jupyter notebook, that's why the flag is `wb`. Please check the `/home/xilinx/jupyter_notebooks/bnn/` folder exists, if not change the path to an existing path.

yyyzzzyyy, 2020-07-28:
> That's a file which is generated in the jupyter notebook, that's why the flag is `wb`. Please check the `/home/xilinx/jupyter_notebooks/bnn/` folder exists, if not change the path to an existing path.

Thank you. I create a new file with the same name and solve the error.

closed 2020-07-28 by giuliogamba