Xilinx/BNN-PYNQ #142: hardware rebuild

maik123456789, 2020-09-08:
Hi! I wanted to rebuild the HW but when I call the make-hw.sh script the following error and I don't know what happens.
[image]

I've looked to other posts with the same issue and didn't suceed. I cloned the repository with the --recursive and also I sourced the vivado 2018.2

¿What else can I try?

maik123456789, 2020-09-08:
And sometimes I get this.
[image]

giuliogamba, 2020-09-09:
You have to set the `XILINX_BNN_ROOT` env variable to the exact locations. Move under `<clone_path>/BNN_PYNQ/bnn/src/`, type `pwd` and use that. It seems you set the variable to `Documentos/`, but you need the absolute path.

closed 2020-09-09 by giuliogamba