Xilinx/BNN-PYNQ #143: vivado_hls: command not found

libby28, 2020-09-16:
As illustrated below, I'm having issues with the command in make-hw.sh

vivado_hls -f $HLS_SCRIPT -tclargs $NETWORK-$PLATFORM $HLS_SRC_DIR %PARAMS %TEST_INPUT $TEST_RESULT $PLATFOR_PART $TARGET_CLOCK
with the error being

"vivado_hls: command not found"

I think this erorr is then what creates all the logged issues as the hls can't run properly.
Given I've added the vivado_hls files etc to path, and it can find them, I'm not sure why it can't then run that command.

Any help would be much appreciated

[image]

giuliogamba, 2020-09-16:
`vivado_hls` has not been sourced properly. Check if you can execute `vivado_hls `in the shell before launching the script. 
In order to properly set-up the toolchain, you need to source the script in the installation path, it should be something like `<installation_path>/installs/lin64/Vivado/2020.1/settings64.sh`

closed 2020-09-16 by giuliogamba

libby28, 2020-09-16:
I sourced it directly in the script, so thought I had got around that issue.
[image]

How would I do that sourcing, not sure which path to add it to

giuliogamba, 2020-09-16:
No, you need to type: 
`source /tools/Xilinx/Vivado/2019.1/installs/lin64/Vivado/2019.1/settings64.sh`

libby28, 2020-09-16:
Thank you so much !
Didn't have too much experience with ubuntu coming into it, so thank you for making it simple for me !