Xilinx/BNN-PYNQ #145: Not able to detect 64-bit Architecture while rebuilding the design

Rajatbansal36786, 2020-09-30:
Hi Sir,

i have installed Vivado HLS design suite 2018.2, on that i am working through WSL to rebuild the design. 
the script make-hw.sh error out that it is not able to detect 64-but architecture [Attaching Snap for ref.].
Please let me know if i have skipped something or what should i do next.
[image]

Rajatbansal36786, 2020-09-30:
i have Vivado installed in Win 10 64-bit.

giuliogamba, 2020-09-30:
I think there's a problem with your Vivado installation and the environment. Have you installed Vivado HLS in your windows machine using the windows installer or in the WSL enviroment? 
This [guide ](https://forums.xilinx.com/t5/Installation-and-Licensing/Installing-Vivado-on-WSL-Windows-Subsystem-for-Linux/td-p/1101914) might be of help.