Xilinx/BNN-PYNQ #147: CASSERT_DATAFLOW condition is not met

mohitajais, 2020-10-28:
Hii,

I have modified the values of num_filters in cnv.py and according to that, trained the network on our dataset with input dimension 64x64x3. Also, changed the values of ifm_ch and ofm_ch in the weights-W1A1.py file. Given here,

ifm = [64, 62, 30, 28, 13, 11]
ofm = [62, 60, 28, 26, 11, 9]
ifm_ch  = [ 3, 6, 16, 32, 32, 64]
ofm_ch  = [ 6, 16, 32, 32, 64, 64]   

Now, I have modified the DoCompute function at the top.cpp file according to the number of filters used. When I run this for bitstream generation, the following warning and error occurs-

WARNING: Hls:: stream 'StreamingConvLayer_Batch.convInp' contains leftover data, which may result in RTL simulation hanging.
CASSERT_DATAFLOW condition is not met 
@E Simulation failed: Function 'main' returns nonzero value '255'.
ERROR: [SIM 211-100] 'csim_design' failed: nonzero return value.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Tue Oct 27 16:06:56 2020...
ERROR: [SIM 211-100] 'csim_design' failed: nonzero return value.
Error in Vivado_HLS

Let me know where I am wrong in all these steps?? Please tell me how to solve this error??

CASSERT_DATAFLOW condition is not met- What does it mean??

giuliogamba, 2020-10-28:
CASSERT_DATAFLOW is basically a check for template values. Some of the HLS functions cannot function correctly if the template values of the instantiation do not follow some guidelines, and this assert is raised. If you open the Vivado HLS GUI, you can check which layer has the condition not met and it's easier to understand which condition has not been met in your design.

mohitajais, 2020-12-09:
Hello,

Thank you,
Yes, the CASSERT_DATAFLOW error is now solved. I have checked the template values and changed it according to my requirements.

closed 2020-12-09 by mohitajais