Xilinx/BNN-PYNQ #150: ERROR: Command errored out with exit status 128: git clone -q https://github.com/Xilinx/BNN-PYNQ.git /tmp/pip-req-build-juh8zblg Check the logs for full command output.

bahadir25, 2021-02-16:
I enter to termina this line "sudo pip3 install git+https://github.com/Xilinx/BNN-PYNQ.git" I get "ERROR: Command errored out with exit status 128: git clone -q https://github.com/Xilinx/BNN-PYNQ.git /tmp/pip-req-build-juh8zblg Check the logs for full command output." 

Please help me

bahadir25, 2021-02-16:
please immediately help

bahadir25, 2021-02-16:
[image]

giuliogamba, 2021-02-17:
I suggest to check the logs for the full command output. What error is raised in that log?

bahadir25, 2021-02-17:
> I suggest to check the logs for the full command output. What error is raised in that log?

I didn't understand what you mean. Can you write more clearly what is a log?

giuliogamba, 2021-02-17:
I mean log file (https://en.wikipedia.org/wiki/Log_file), as stated in the error you should check the generated log file. 
p.s. are you sure the board is connected to internet?

bahadir25, 2021-02-17:
So how can I check from where? Please help me

bahadir25, 2021-02-17:
I performed earlier such as project in link https://www.youtube.com/watch?v=LoLCtSzj9BU&t=724s . It run

giuliogamba, 2021-02-17:
You might want to read the PYNQ getting started https://pynq.readthedocs.io/en/v2.6.1/getting_started/pynq_z1_setup.html. 
If the board is plugged to your laptop, most likely it's not connected to the internet. You should connect it to a router in order to have the board connected to the internet.

closed 2021-02-17 by giuliogamba