Xilinx/BNN-PYNQ #151: CSim failed with errors

mohitajais, 2021-03-22:
Hello,

I have trained a new network on my own dataset having input dimensions of 64x64x3 and with given topology
ifm  = [64, 62, 30, 28, 13, 11]
ofm  = [62, 60, 28, 26, 11, 9]   
ifm_ch  = [ 3, 32, 32, 64, 64, 128]
ofm_ch  = [ 32, 32, 64, 64, 128, 128]   
According to this, I have modified top.cpp and main_python.cpp file also. When I run this file, the following error occurs

Setting network weights and thresholds in accelerator...
terminate called after throwing an instance of 'char const*'
@E Simulation failed.
ERROR: [SIM 211-100] CSim failed with errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source /home/lnmiit/BNN-PYNQ/bnn/src/network/hls-syn.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Mon Mar 22 10:33:11 2021...
ERROR: [SIM 211-100] CSim failed with errors.
Error in Vivado_HLS

Please tell me how to solve this error??
terminate called after throwing an instance of 'char const*' - What does it mean??

giuliogamba, 2021-03-22:
Hi, 

did you also export the binary files for the weights and thresholds and passed to the csim the proper pointer to the folder with the newly generated parameters and updated the `config.h` file in the hardware folder?
That error is raised when there's been a problem reading those binary files (as in [here](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/cnvW1A1/sw/main_python.cpp#L71)).

mohitajais, 2021-03-23:
Hi,

Yes, I have updated the config.h file in hardware folder and also placed newly generated binary files for weights in params folder. Still getting the same error.

giuliogamba, 2021-04-02:
Can you investigate if the binary files are properly read?

mohitajais, 2021-04-06:
Hii.

Yes, I have checked the binary files are properly updated in params folder. I have a question regarding the input dimensions i.e 64x64x3 that I have used in my network topology. Do I need to modify the dimensions in below-given line of foldedmv-offload.h file .

tiny_cnn::chaninterleave_layer<tiny_cnn::activation::identity> interleaver(3, 32 * 32, false);

Instead of 32*32, Should I have to write 64*64. Maybe due to this it is showing the error of terminate called after throwing an instance of 'char const*' during the CSim step.

mohitajais, 2021-04-06:
Hii.

Yes, I have checked the binary files are properly updated in params folder. I have a question regarding the input dimensions i.e 64x64x3 that I have used in my network topology. Do I need to modify the dimensions in below-given line of foldedmv-offload.h file .

tiny_cnn::chaninterleave_layer<tiny_cnn::activation::identity> interleaver(3, 32 * 32, false);

Instead of 32 * 32, Should I have to write 64 * 64. Maybe due to this it is showing the error of terminate called after throwing an instance of 'char const*' during the CSim step.

giuliogamba, 2021-04-12:
Yes, that could be a possible reason, can you try to execute the step-by-step flow in Vivado HLS and double check which function fails?

mohitajais, 2021-04-22:
Hii,
Yes, I have tried to execute step by step flow in vivado HLS, & the problem is solved now by modifying only line 298 in foldedmv-offload.h file by replacing 32x32 dimension with 64x64. The bitstream file has now succesfully generated.

Thanks Giuliogamba!

closed 2021-04-22 by mohitajais