Xilinx/BNN-PYNQ #152: edit tcl file

mustafatasci42, 2021-04-01:
how can open pynq tcl file in vivado. I tried to open it with vivado source command, it gives an error . vivado version 2019.1 installed on ubuntu , 2018.2 installed on win10.

giuliogamba, 2021-04-02:
Hi, 
I'm not sure about what you are trying to do and what kind of error you are seeing, can you provide more details? In general, you can see the build flow in the [make-hw.sh](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-hw.sh) script, which is first executing Vivado HLS at line [145 ](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-hw.sh#L145) and after Vivado for the bitstream generation at line [184](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/make-hw.sh#L184).

mustafatasci42, 2021-04-03:
I want to open cnvW1A1-pynqZ1-Z2.tcl file in vivado2018.2. To work on hardware developed in fpga. 
[image]

giuliogamba, 2021-04-06:
Hi, the tcl script is meant for automatic execution, since it is called from the `make-hw.sh` script. 
If you want to modify the hardware design, I would suggest to execute the [hardware design rebuild](https://github.com/Xilinx/BNN-PYNQ#hardware-design-rebuilt) flow, with `make-hw.sh` with the `-a` flag, and then open the generated vivado project, which will be written under `<clone_path>/BNN_PYNQ/bnn/src/network/output/vivado`.

mustafatasci42, 2021-04-09:
Hi giuliogamba. Thanks for your responses. I try this way but I am getting the error in the picture. Do I need to specify another env? 
[image]
[image]

giuliogamba, 2021-04-12:
Hi, I think there's a problem with your installation or set-up of Vivado. Please refer to [this ](https://forums.xilinx.com/t5/High-Level-Synthesis-HLS/ERROR-TECH-200-14/m-p/999200) answer on Xilinx forum about the language settings.

mustafatasci42, 2021-11-30:
Thanks for the reply. The error is caused by the ubuntu installation language.

closed 2021-11-30 by mustafatasci42