Xilinx/BNN-PYNQ #153: csim error

rkarn, 2021-04-07:
I run the command:    ./make-hw.sh lfcW1A2 pynqZ1-Z2 h

I got below error. I am running it on linux centos machine with vivado 2020.2 

INFO: [SIM 211-1] CSim done with 0 errors.
INFO: [SIM 211-3] *************** CSIM finish ***************
ERROR: [HLS 200-644] The 'csim_design -compiler' hidden command is not supported.
INFO: [HLS 200-111] Finished Command csim_design CPU user time: 10.22 seconds. CPU system time: 0.67 seconds. Elapsed time: 9.42 seconds; current allocated memory: 66.645 MB.
command 'csim_design' returned error code
    while executing
"csim_design -argv "$directory_params $test_image 10 $expected_result" -compiler clang"
    (file "/root/BNN-PYNQ/bnn/src/network/hls-syn.tcl" line 79)
    invoked from within
"source /root/BNN-PYNQ/bnn/src/network/hls-syn.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

giuliogamba, 2021-04-12:
Can you try by removing the `-compiler clang` from line 79 of [hls-syn.tcl](https://github.com/Xilinx/BNN-PYNQ/blob/master/bnn/src/network/hls-syn.tcl#L79) file?

rkarn, 2021-04-13:
#### I run by removing the "--compiler clang" option from line 79.  I get below output :
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch GCC as the compiler.
   Compiling ../../../../../../../library/host/rawhls-offload.cpp in debug mode
   Compiling ../../../../../../../library/host/foldedmv-offload.cpp in debug mode
   Compiling ../../../../../../lfcW1A2/sw/main_python.cpp in debug mode
   Compiling ../../../../../../lfcW1A2/hw/top.cpp in debug mode
   Generating csim.exe

#### Up to there its fine. Then it has generated below errors:

ERROR: [HLS 214-134] in function 'WidthAdjustedInputStream<64u, 64u, 13u>::WidthAdjustedInputStream(hls::stream<ap_uint<64>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:735:85)
ERROR: [HLS 214-134] in function 'WidthAdjustedOutputStream<64u, 64u, 32u>::WidthAdjustedOutputStream(hls::stream<ap_uint<64>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:768:7)
ERROR: [HLS 214-134] in function 'WidthAdjustedInputStream<64u, 64u, 32u>::WidthAdjustedInputStream(hls::stream<ap_uint<64>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:735:85)
ERROR: [HLS 214-134] in function 'WidthAdjustedOutputStream<128u, 128u, 16u>::WidthAdjustedOutputStream(hls::stream<ap_uint<128>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:768:7)
ERROR: [HLS 214-134] in function 'WidthAdjustedInputStream<128u, 128u, 16u>::WidthAdjustedInputStream(hls::stream<ap_uint<128>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:735:85)
ERROR: [HLS 214-134] Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:752:0)
ERROR: [HLS 214-134] in function 'WidthAdjustedOutputStream<16u, 64u, 4u>::WidthAdjustedOutputStream(hls::stream<ap_uint<64>, 0>&, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/streamtools.h:752:87)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:108:60)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:109:59)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:108:60)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:109:59)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:108:60)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:109:59)
ERROR: [HLS 214-134] in function 'DoCompute(ap_uint<64>*, ap_uint<64>*, unsigned int)': Pointer to pointer is not supported (/root/BNN-PYNQ/bnn/src/library/finn-hlslib/fclayer.h:109:59)
ERROR: [HLS 214-135] Syn check fail!
INFO: [HLS 200-111] Finished Command csynth_design CPU user time: 11.68 seconds. CPU system time: 0.8 seconds. Elapsed time: 14.53 seconds; current allocated memory: 90.104 MB.
Pre-synthesis failed.
    while executing
"source /root/BNN-PYNQ/bnn/src/network/hls-syn.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total CPU user time: 22.2 seconds. Total CPU system time: 1.85 seconds. Total elapsed time: 24.83 seconds; peak allocated memory: 89.753 MB.
INFO: [Common 17-206] Exiting vitis_hls at Tue Apr 13 06:54:51 2021...

giuliogamba, 2021-04-13:
Have you modified the script to execute `vitis_hls` instead of `vivado_hls`?

rkarn, 2021-04-13:
Yes, I made changes at two places in make-hw.sh..

Line 59  :   PATH_TO_VIVADO_HLS=$(which vitis_hls)
Line 145:   vitis_hls -f $HLS_SCRIPT -tclargs $NETWORK-$PLATFORM $HLS_SRC_DIR $PARAMS $TEST_INPUT $TEST_RESULT $PLATFORM_PART $TARGET_CLOCK

giuliogamba, 2021-04-14:
I would strongly suggest moving back to `vivado_hls`, given all the flow has been validated with that tool (so installing 2020.1 or earlier versions)
If you need to use `vitis_hls`, you should change the branch of the `finn-hlslib` under `bnn/src/library/` to `vitis_hls`.

closed 2021-05-26 by giuliogamba