Xilinx/BNN-PYNQ #154: Cant find library and locate the .so file

Johsamue, 2021-04-26:
Hi, I have an PYNQ image v2.4 and have cloned this repo, but cant run the 3thed step:

I get this error but when I go there I can see the file and its executable, do I need to set a path or how do I solve the error? 

OSError: cannot load library '/usr/local/lib/python3.6/dist-packages/bnn/libraries/pynqZ1-Z2/python_hw-cnvW1A1-pynqZ1-Z2.so': /usr/local/lib/python3.6/dist-packages/bnn/libraries/pynqZ1-Z2/python_hw-cnvW1A1-pynqZ1-Z2.so: undefined symbol: cma_alloc.  Additionally, ctypes.util.find_library() did not manage to locate a library called '/usr/local/lib/python3.6/dist-packages/bnn/libraries/pynqZ1-Z2/python_hw-cnvW1A1-pynqZ1-Z2.so'

giuliogamba, 2021-04-26:
Hi, on which board are you executing this?

Johsamue, 2021-04-26:
On a Pynq Z2 board

giuliogamba, 2021-04-26:
Did you install the package using pip (as in the [quick start](https://github.com/Xilinx/BNN-PYNQ#quick-start)) or just cloned the repo? In order to execute the available jupyter notebooks on the board you should install the package.

Johsamue, 2021-04-27:
Yes I followed the quick start and did pip install, found bnn in my jupiter notebook.

giuliogamba, 2021-04-27:
Are you executing one of the jupyter notebooks or you're experiencing this issue in new code? If it's in a new codebase, are you importing the bnn library (`import bnn`) and experiencing the error during import?