Xilinx/BNN-PYNQ #156: tutor for CNV-BNN_Road-Signs?

goldinoragung, 2021-05-25:
Hy Everyone

i had some issues when i tried image classification, i tried to classified a car pict with 720p. when i tried at line "Here we launch the classification on all the tiles from the source image" and make stride s//2 and it shows 1144. at first result, there's no bounding box in it even it classified as a automobile. and i tried the next line and error and it says "cannot reshape array of size 11440 into shape"

goldinoragung, 2021-05-27:
and i tried to re-run the road signs with same picture and there's no box in it. is it possible that actually error?

giuliogamba, 2021-06-01:
It is possible, the neural network does not have an 100% accuracy. It is important nonetheless that the tiling of the original image is performed correctly. This means making sure that the crop of the original image is done such that one or more tiles completely embed the object you want to recognize.

goldinoragung, 2021-06-15:
Hi Mr. giuliogamba,
Now i know what should i do, i appreciate with your answer and i'm very happy to discuss it again with you.
Thank You So Much

closed 2021-06-16 by giuliogamba