Xilinx/BNN-PYNQ #162: Images per second

Ronin3502, 2021-11-26:
Hi, I noticed that, in LFC-QNN_MNIST.ipynb, the results of case 1 and case 2 are both 'Inference took 28040.00 microseconds, 2.80 usec per image', which indicated there are 10,000,000 testing images, but the size of MNIST test dataset is 10,000. I cannot find the C/C++ code of python_hw-lfcW1A1-pynqZ1-Z2.so, so I failed to find the implementation of the inference function.
It would be so kind if you could explain this. Thanks.
[image]

nickfraser, 2021-11-26:
I think you've misread the units here (**microseconds**, not milliseconds!):
 - 1 microsecond = 1 usec
 - 28,040us / 10,000 = 2.8us

You are correct, there are 10,000 test images.

closed 2021-11-26 by nickfraser

Ronin3502, 2021-11-26:
Oh, my bad.
I got it. Thanks a lot!