Xilinx/BNN-PYNQ #163: [SIM 211-100] 'csim_design' failed: compilation error(s).   rebuilding error

mustafatasci42, 2021-11-30:
**Hi, This issue was created by someone else before but it wasn't answered. I was hoping, if you could resolve this.** 

mustafa@mustafat:~/Documents/BNN-PYNQ/bnn/src/network$ ./make-hw.sh lfcW1A1 pynqZ1-Z2 a
xilinx-tiny-cnn already cloned
FINN hls library already cloned
Calling Vivado HLS for hardware synthesis...

****** Vivado(TM) HLS - High-Level Synthesis from C, C++ and SystemC v2018.3 (64-bit)
  **** SW Build 2405991 on Thu Dec  6 23:36:41 MST 2018
  **** IP Build 2404404 on Fri Dec  7 01:43:56 MST 2018
    ** Copyright 1986-2018 Xilinx, Inc. All Rights Reserved.

source /tools/Xilinx/Vivado/2018.3/scripts/vivado_hls/hls.tcl -notrace
INFO: [HLS 200-10] Running '/tools/Xilinx/Vivado/2018.3/bin/unwrapped/lnx64.o/vivado_hls'
INFO: [HLS 200-10] For user 'mustafa' on host 'mustafat' (Linux_x86_64 version 5.4.0-90-generic) on Tue Nov 30 22:05:45 +03 2021
INFO: [HLS 200-10] On os Ubuntu 18.04.2 LTS
INFO: [HLS 200-10] In directory '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn'
HLS project: lfcW1A1-pynqZ1-Z2
HW source dir: /home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw
BNN HLS library: /home/mustafa/Documents/BNN-PYNQ/bnn/src/library/finn-hlslib
INFO: [HLS 200-10] Opening project '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2'.
INFO: [HLS 200-10] Adding design file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw/top.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/lfcW1A1/hw/../sw/main_python.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/library/host/foldedmv-offload.cpp' to the project
INFO: [HLS 200-10] Adding test bench file '/home/mustafa/Documents/BNN-PYNQ/bnn/src/library/host/rawhls-offload.cpp' to the project
INFO: [HLS 200-10] Opening solution '/home/mustafa/Documents/BNN-PYNQ/bnn/src/network/output/hls-syn/lfcW1A1-pynqZ1-Z2/sol1'.
INFO: [SYN 201-201] Setting up clock 'default' with a period of 5ns.
INFO: [HLS 200-10] Setting target device to 'xc7z020clg400-1'
INFO: [SIM 211-2] *************** CSIM start ***************
INFO: [SIM 211-4] CSIM will launch CLANG as the compiler.
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
INFO: [SIM 211-3] *************** CSIM finish ***************
4
    while executing
"source [lindex $::argv 1] "
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 { source [lindex $::argv 1] } "

INFO: [Common 17-206] Exiting vivado_hls at Tue Nov 30 22:05:46 2021...
ERROR: [SIM 211-100] 'csim_design' failed: compilation error(s).
Error in Vivado_HLS

zGitHubj, 2021-12-19:
Hello, did you solve this problem? i encounter the same problem as this one.

mustafatasci42, 2021-12-21:
yes my problem is solved. the problem was caused by the ubuntu installation language. ubuntu installation language must be english

zGitHubj, 2021-12-21:
BEST SOLUTION
Hello,
I have seen the same issue with ubuntu 18.4 and Vivado 19.1.
I found this answer:
https://forums.xilinx.com/t5/Design-Entry/Vivado-2018-3-error-launch-hls/td-p/940856
So, after installing build-essential the issue gone.

url: https://support.xilinx.com/s/question/0D52E00006hplQGSAY/hls-20183-csim-is-broken-on-ubuntu-1804-lts?language=en_US

closed 2022-01-04 by nickfraser